xjtuecho/EBAZ4205 #3: KG4205 Development board VGA &UART

firepower9966, 2020-12-05:
[image]

Do you have any info on KG4205 Which match the EBAZ4205.  It provides VGA and UART and a 2.54mm  I/O connector which can attach optional camera.

xjtuecho, 2020-12-09:
This KG4205 is an extended card of EBAZ4205. I didn't know more about it, but  I can purchase it easily in my country.

ryanm101, 2020-12-20:
Seems you can buy a few adaptors: https://www.aliexpress.com/item/1005001530233006.html?spm=2114.12057483.0.0.61c9febeBUtvsg

Where can you buy the KG4205 from?

firepower9966, 2020-12-20:
US $17.69  9%OFF | ZYNQ7010 Development Board, Xc7z010 FPGA.
https://a.aliexpress.com/_msejyAv

artemglukhov, 2021-02-01:
May I ask you if you know the part number of these 2x10 20mm connectors?

firepower9966, 2021-02-01:
Search for
20 Position 2 mm 2 Row Headers & Wire Housings
Depends if you want wired connector or PCB mount.

BrainHunter, 2021-03-20:
does anyone have schematics for this board?

nic3-14159, 2021-03-20:
> does anyone have schematics for this board?

There's some info about this board in the discussions
https://github.com/xjtuecho/EBAZ4205/discussions/46