xjtuecho/EBAZ4205 #4: Information on readme how to upload bitstream to FPGA?

Kazhuu, 2020-12-11:
Hello! I'm relative new to Xilinx FPGAs and ordered one of these boards. I'm wondering what is the best way to actually do some FPGA development on this board? I think I can put the bitstream file to SD card and let linux upload it to FPGA on boot but cannot find any information about this. Also someone on Reddit mentioned that you need JTAG FPGA programming cable from Xilinx. I guess this would be the best way to do the development? But the cable itself cost something like 80€ I think.

I've used Altera's development boards earlier which can be used with USB cable instead. Thank you!

xjtuecho, 2020-12-11:
Maybe you need a Digilent JTAG-SMT2，the FT232HL_v20.5.31.pdf in this repo is enough. cost about 8$

Kazhuu, 2020-12-11:
Thanks! With this information I was actually able to do some research and found web page describing [programming solutions](https://store.digilentinc.com/jtag-programmers-fpga-programming-solutions/). I'll also had a look at FT232HL board you mentioned and was thinking could [this](https://www.adafruit.com/product/2264) Adafruit one serve similar purpose? Of course with this you have to connect the pins individually.

tpecar, 2020-12-12:
Hi, [I'm probably from the same bandwagon as @Kazhuu](https://hackaday.com/2020/12/10/a-xilinx-zynq-linux-fpga-board-for-under-20-the-windfall-of-decommissioned-crypto-mining/).

With a bunch of these on the way I'm also in the process of figuring out how to use them :rofl: 

The cheapest way to get going is probably to just build the SD card image and boot the board from that. Most examples available for this board go this way
- https://www.jianshu.com/p/b83c663ecaaa
- https://gitee.com/actionchen/ebaz4205_hw/blob/master/Doc/ebaz4205_introduce.md
- https://www.youtube.com/watch?v=b6g5tNZZnVc

However, in order to debug a baremetal application running on PS (ARM core) and to not loose sanity swapping SD cards, I also want to figure out how to use JTAG.

#### The easy and expensive way of doing this
The [Digitlent JTAG cables](https://store.digilentinc.com/jtag-programmers-fpga-programming-solutions/) are the official cables that are supported by Xilinx tools. If you want the minimum amount of fuss, get one of these.

However, given that they are only glorified FTDI dongles programmed with configuration blessed by Xilinx, it makes me a bit mad spending $60 for one of these.

#### The cheap and finnicky way of doing this
While I couldn't find a product that uses the exact design specified in FT232HL_v20.5.31.pdf, I suppose that most generic FTDI JTAG dongles are applicable here.

You can then use [OpenOCD](http://openocd.org/) or [xc3sprog](http://xc3sprog.sourceforge.net/) to download the bitstream to the fpga (provided that you have boot mode pins set to JTAG - these are the same pins @xjtuecho suggests to change to get SDcard boot).

There's a couple of gotchas however:
- xc3sprog is only capable of setting up the PL (FPGA) side, PS (ARM core) is uninitialized
- OpenOCD has support to set up PL / PS sections
  - [OpenOCD page](https://devel.rtems.org/wiki/Debugging/OpenOCD/Xilinx_Zynq) has configuration files for Zynq but it needs to be adapted to your specific chip
  - [Zynq-RS project](https://git.m-labs.hk/M-Labs/zynq-rs) has configuration files set up for Z7-10

  but is not supported by Xilinx, so there's probably some effort required to get a decent debugging environment
- none of these metods give you access to [ILA](https://www.xilinx.com/video/hardware/logic-debug-in-vivado.html)

Then you have _Xilinx Virtual Cable (XVC)_.
Seems to be the most interesting method. Instead of talking to the debug probe directly, the Xilinx tools talk to a XVC deamon over TCP/IP, which can run on your system or some other device, which is actually connected to your target
- [original XVC deamon](https://debugmo.de/2012/02/xvcd-the-xilinx-virtual-cable-daemon/) and [its sister project](https://code.google.com/archive/p/playtag/) run on your development system and talk to the target via FTDI2232 (essentially bypassing the silly restrictions). However, these projects seem to be inactive.
- [Xilinx has reworked this code to run directly on a Zynq platform](https://github.com/Xilinx/XilinxVirtualCable), so if you have more than one board, you can use one as a debug probe.
- [somebody has ported the Xilinx example on the Raspberry Pi](https://github.com/derekmulcahy/xvcpi) so you can try that.

Kazhuu, 2020-12-12:
@tpecar thanks for the very detailed answer! This really helps people like me to get started and learning. Very appreciated! I think I might end up buying a cable from Digilent because it can also be used in the future with different boards. And I'm interested in the debugging support which is quite mandatory when doing anything more complicated.

One thing I'm wondering. If I buy JTAG-HS3 instead of JTAG-HS2. HS3 has support for PS_SRST pin resetting the processor during debugging. Although I checked the schematics of this board and it seems this pin is not connected to JTAG header. I'm guessing if you want PS_SRST pin support then you'd need to modify the board.  Here is good explanation of [PS_SRST](https://forum.digilentinc.com/topic/580-jtag-hs3-vs-jtag-hs2-when-using-zynq-with-vivado/).

tpecar, 2020-12-12:
Well, given that they sell both JTAG-HS2 and JTAG-HS3 for the same price, I guess you select the one that better fits your use case.

HS2 supports [2-Wire JTAG](https://sudonull.com/post/106128-We-disassemble-the-2-wire-JTAG-protocol), but since we already have standard JTAG routed out, this is a non-issue.

HS3, as you mentioned, has an open drain output which the Xilinx tools can use for PS reset.

The PS_SRST_B (B10 on CLG400 package) is routed out to R2430 (pullup to VCC) which is perfect - just run a bodge wire to the JTAG connector (pin 12, 14 of the connector seem to be disconnected).

[image]

[image]

Kazhuu, 2020-12-13:
Thanks you for you detailed answer! I'll order one and try it out. Sorry might be beginner question but what I understood that HS3 has open drain outputs, meaning that I have to connect it to VCC pull up like you said. So in this case I need to connect from the **left side** of the resistor to pin 14 like so.

[image]

tpecar, 2020-12-13:
Yep, that should work.

It obviously needs to be tested, and I probably have a good 2 months before I get the boards, so if you can try that earlier I'll appreciate your report.

Once it gets tested, we can consider integrating it into the readme.

Kazhuu, 2020-12-13:
Sounds good. I can test it but it might not be very soon because I'm not at home during Christmas holidays and getting that cable to Finland will take some time delivered. I'll report here when I've tested this.

tpecar, 2020-12-13:
> getting that cable to Finland will take some time delivered

Just in case you didn't notice, Digilent has European distributors - you can get the cable at [Farnell for example](https://analytics.oemsecrets.com/main.php?p=410-299&m=Digilent&q=117&n=Farnell%20element14%20SI&table=api&media=buynow&source=digilent&event_link=https%3A%2F%2Fsi.farnell.com%2F2490176%3FCMP%3DGRHB-OEMSECRETS).

ustcpetergu, 2020-12-16:
Actually, you can download the bitstream over ethernet in u-boot. So no need to solder SD card slot or buy expensive JTAG downloaders. UART is still needed though. 

Set up a tftp server on host computer and in u-boot use command similar to this: 
```
<press d>
setenv serverip 192.168.6.6
setenv ipaddr 192.168.6.136
tftpboot 0x100000 bd_hello_wrapper.bit.bin
fpga load 0 0x100000 0x1fcba0

reset
```

Bitstreams must be converted to bin files to run, use this tcl script: 
```
proc gen_bin {bit} {
    set filename [file rootname [file tail $bit]].bif
    set fileId [open $filename "w"]
    puts $fileId "all:"
    puts $fileId "{"
    puts $fileId "  $bit"
    puts $fileId "}"
    close $fileId
    #exec bootgen -w on -image $filename -arch zynqmp -process_bitstream bin
    exec bootgen -w on -image $filename -arch zynq -process_bitstream bin
    puts "$filename created successfully"
}
```

Also, as this board's ethernet is connected to PL instead of PS, the ethernet connection will fail after downloading bitstream, which is quite annoying. So EMIO and the PS must be specially configured(even if not used) to keep net connection working after programming. 

Somewhat detailed instruction can be found in this article on my blog(at the end is the ethernet part): 
https://www.ustcpetergu.com/MyBlog/experience/2020/07/01/ebaz-4205.html
It's in Chinese but I ran a google translation test and it seems OK.

ustcpetergu, 2020-12-16:
And about JTAG downloader, I did by one later. But of course not the expensive official one. I just picked a CJMCU FT232HL(the one with purple PCB, which is quite common) module and flashed the EEPROM. Similar to this: https://gist.github.com/rikka0w0/24b58b54473227502fa0334bbe75c3c1

The firmware between FT232 and FT2232 may be a little different so the gist may not work out-of-the-box, but I got it working at last. Only cost me 30RMB.

tpecar, 2020-12-16:
Thank you @ustcpetergu for the info on downloading bitstreams over Ethernet. I suppose this uses the u-boot of the original miner firmware which is programmed on the flash chip.

Regarding JTAG dongles, it is possible to flash a generic dongle with Digilent firmware. This is against their license, which isn't that much of a problem in case of hobby / personal usage, but it might become important in other settings.

Regarding FT232, correct me if I'm wrong, but I thought you required FT2232 due to MPSSE (which has JTAG capability), at least for interfacing with original Xilinx tools. Other unofficial tools which use FT232 do so by bit-banging the interface.

ustcpetergu, 2020-12-17:
As far as I know FT232H works too. My Vivado detected my FT232H and successfully recognized the 7010 chip when ebaz4205 is connected. I haven't tried anything like bitstream downloading or ILA though(I trashed my board when trying to remove the BGA xc7z010 chip, so won't have any chance to test). 

But as better evidence, the 7010 board by Alinx does use FT232HL as their JTAG chip. I don't have this board but on their user guide, the schematic on page 12 shows so. 
http://www.alinx.vip:81/ug/AX7020_UG.pdf 

One difference I know is that FT2232H is dual-channel so you can have JTAG and USB-to-UART together just using one USB cable, like on Digilent board ZYBO or PYNQ. FT232H is a single-channel one so another standalone USB-to-UART chip is required, in the Alinx case, CP2102 is used.

tpecar, 2020-12-17:
Ok, thank you very much for the clarification!

And you're right, FT232H (which is completely different from FT232R, so suffixes are important) has MPSSE. So, provided you configure its EEPROM, it should work with Xilinx tools.

I guess that this is the method that @xjtuecho has referenced initially, so we've come full circle :rofl: 

Thank you for also providing a concrete example.

ryanm101, 2020-12-20:
There are some good tutorials here: https://www.programmersought.com/article/78153940676/

https://www.programmersought.com/article/65574959208/

HDMI: https://www.programmersought.com/article/80046040945/

tpecar, 2020-12-20:
> Bonus points if anyone actually manages to download and reupload anything from that CSDN site

I feel your pain :laughing: Most chinese forums have some sort of paywall, and some are downright impossible to register into.

I'll take a look during the week if I can get access to some of the sites.

ryanm101, 2020-12-20:
This article is good, I'm thinking a 2-way DPST dip switch: https://www.programmersought.com/article/63646026059/ attach the resistor to the common pole.

R2578 NAND -> R2585 JTAG
R2584 NAND -> R2577 SDCARD

This would let you switch between JTAG for Dev, SDCard and NAND without lots of re-soldering. 
I think (but untested) the truth table would be:
|----|2578|R2585|2584|2577|
|----|-----|------|-----|-----|
|NAND|1|0|0|1|
|SDCARD|1|0|1|0|
|JTAG|0|1|0|1|

Kazhuu, 2021-01-09:
It seems I cannot test what I promised. At least not for now. I've been playing around with the board but I underestimated the amount work it involves. I'm still quite beginner with FPGA development and especially Xilinx tools. I managed to get buildroot linux image working. Then I tried very simple FPGA design to drive one of the pins high but couldn't get it to work for some reason by downloading the bitstream over JTAG. I even tried changing the resistor configuration to sd card and nand flash but no leds turn on or data port pins went high. In the end I also managed to brake my board by braking SD card slot -_-'. 

So at the moment I just felt giving up with this and invest on proper evaluation board with good documentation to get started. Hopefully someone else can test what reset pin functionality. Any recommendations or guides to Xilinx tools with some good FPGA board for beginners?

ustcpetergu, 2021-01-11:
I'm also a new FPGA player and it's true that this board is not beginner-friendly. 
For pure FPGA development actually you can just ignore the ARM cores and using Vivado only, just like on normal 7-series FPGAs. Just driving a pin high don't need clock so things can be a lot easier(because the board don't have clock input on PL side). Since you are downloading bitstream over JTAG, I think Vivado's hardware manager can always be used to program the device no matter what the boot option the resistors mean.

ustcpetergu, 2021-01-11:
And about boards... I think Digilent's boards are good, haven't used any evaluation board yet. Arty seems popular for pure FPGA and PYNQ for ZYNQ series. Though most of the time I do pure FPGA dev, I got a PYNQ for it's relatively cheap price and the abundant(85K) logic resources.

xjtuecho, 2021-01-11:
> I'm also a new FPGA player and it's true that this board is not beginner-friendly.
> For pure FPGA development actually you can just ignore the ARM cores and using Vivado only, just like on normal [...]

For FPGA beginner，Sipeed Tang Nano with builtin programmer maybe a better choice.

ustcpetergu, 2021-01-11:
Yes, my suggestion's only valid if you want to use Vivado(our school teaches that), which is, actually not that good.

tpecar, 2021-01-12:
@Kazhuu No problem, I'll see if I can test this.

Regarding other development boards

- there is a section in the Wiki listing other cheap development boards with their introductory development resources, but it's still in the works
  https://github.com/xjtuecho/EBAZ4205/wiki#beyond-ebaz

- Additionally, a really comprehensive list of cheap FPGAs is available on
https://www.joelw.id.au/FPGA/CheapFPGADevelopmentBoards

Digilent / Trenz Electronic development boards are probably the most documented (but usually also the most expensive) Xilinx development bords.

I'm not up to speed with other manufacturers, but it's probably a good idea that you check what Lattice, Intel/Altera have to offer.

You should also check for boards that are supported by the yosys / SymbiFlow open source HDL synthesis tools - people usually target cheap boards there.

xjtuecho, 2021-01-28:
> @Kazhuu No problem, I'll see if I can test this.
>
> Regarding other development boards
>
> * there is a section in the Wiki listing other cheap development boards with their introductory development [...]
What's the version of you KiCad? I can't open the KiCad project with latest KiCad 5.1.9-1

tpecar, 2021-01-28:
I'm using nightly build of kicad (compiled from current git master), it's numbered as version 5.99.

I've needed to use nightly because the have significantly improved the Altium PCB importer and they have switched to s-expr file format for eeschema, for which it was easier to write a schematic generation tool.

If you're on Windows, CERN provides prebuilt binaries (see bottom of page for latest version)
https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/nightly/

For linux it depends on the distribution - for Ubuntu/Debian, the instructions are provided on the Kicad page
https://kicad.org/download/ubuntu/#_nightly_development_builds

M3m3M4n, 2021-09-27:
> And about JTAG downloader, I did by one later. But of course not the expensive official one. I just picked a CJMCU FT232HL(the one with purple PCB, which is quite common) module and flashed the [...]

@regymm Does the firmware need to be modified? can you share the binary?

regymm, 2021-09-27:
For FT2232 just follow the link. You can try this one(use xxd to reverse to binary) for FT232, and change the product_id and file name in the link-provided configuration file accordingly. Run `dmesg --follow` and re-plug your downloader to see whether the firmware is successfully flashed. May need multiple tries but this is not that hard. 
```
00000000: 0100 0304 1460 0009 8000 0800 0000 a012  .....`..........
00000010: b228 da1a 0000 0000 0000 0000 0000 5600  .(............V.
00000020: 0100 c792 6a35 5401 0031 4a74 6167 536d  ....j5T..1JtagSm
00000030: 7432 0000 0000 0000 0000 0044 6967 696c  t2.........Digil
00000040: 656e 7420 4a54 4147 2d53 4d54 3200 0000  ent JTAG-SMT2...
00000050: 0000 0000 0000 0000 1300 0000 0000 0000  ................
00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000080: 0000 0000 0000 0000 0000 4800 0000 0000  ..........H.....
00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000a0: 1203 4400 6900 6700 6900 6c00 6500 6e00  ..D.i.g.i.l.e.n.
000000b0: 7400 2803 4400 6900 6700 6900 6c00 6500  t.(.D.i.g.i.l.e.
000000c0: 6e00 7400 2000 5500 5300 4200 2000 4400  n.t. .U.S.B. .D.
000000d0: 6500 7600 6900 6300 6500 1a03 3200 3100  e.v.i.c.e...2.1.
000000e0: 3000 3200 3500 3100 4100 3000 3800 3800  0.2.5.1.A.0.8.8.
000000f0: 3700 3000 0203 0000 0000 0000 0000 9887  7.0.............
```

M3m3M4n, 2021-09-27:
Thanks.
Also earlier I missed a comment on https://gist.github.com/rikka0w0/24b58b54473227502fa0334bbe75c3c1 that mentions the HS2 firmware is confirmed working with 232h.

AndersBNielsen, 2021-12-10:
I have a FT232 coming in the mail but to anyone that has a Raspberry Pi around: xvcpi was suuuuper easy to use. 
Simply connect it to the "SPI" port on the RPi + GPIO25 and you can connect through Vivado. Super handy with a Pi Zero W - Wireless JTAG :)

regymm, 2021-12-10:
Yeah, RPi's neat. How long does it take to program ZYNQ 7010? I'd like to use RPi but seems on my side it takes more than half a minute and when using ILA to capture data it's very slow.

nic3-14159, 2021-12-12:
@regymm  The xvcpi code, built without changes, programs my board in about 20s. Lowering the clock delay as much as I could, I was able to get it down to about 8s. However, there seems to be a timing issue with the code which limits the minimum clock delay/maximum clock rate, after fixing that (I have a pull request open in the original repo) I was able to get the programming speed down to 3s after setting the clock delay to 0.