xjtuecho/EBAZ4205 #5: PcbSch for EBAZ4205.PcbDoc

tpecar, 2020-12-12:
Do you have access to the PcbSch (schematic file) of the design?

We can obviously make do with the pdf export, but the actual design file would be welcome so that we can utilize cross-probing in Altium.

tpecar, 2020-12-12:
Also, correct me if I'm wrong, but it seems that some traces / planes are missing in the PCB design?

Notably, some VCC lines (I've looked at R2430) don't seem to go anywhere.

xjtuecho, 2020-12-13:
sorry, no .schdoc. The pdf schematic and PcbDoc are probably not the original design. Maybe they are the product of some reverse engeering.

tpecar, 2020-12-13:
Ok, thank you. The design files probably originate from the initial design, but it seems that someone has removed a couple of nets while this design was passed around.

It seems that the original reseller of this board on AliExpress was aware of such versions, noting "PCB (ad format copy board diagram, you can only refer to it, it is not recommended to proof)".

In the meantime I've found a more complete PcbDoc at
https://github.com/Elrori/EBAZ4205

I'll contact him to see if he has any more info.

tpecar, 2020-12-19:
Well, I'm currently still waiting for an answer from Elrori, but I did some digging of my own in the meantime.

This board was making rounds on chinese EE forums in the early/mid 2019, and the relevant development resources have been reposted many times over. We're probably well past the point where anyone affected would care about further distributing these files, so the least we can do is organize them.

@xjtuecho I suggest that you have a look at Elrori's version of the pcbdoc and replace your version with his.

tpecar, 2020-12-19:
Here's a couple forum threads from when the boards initially hit the chinese flea market.
I didn't yet bother to register on them, so I didn't check the files.

- http://bbs.eeworld.com.cn/thread-1078942-1-1.html
- http://bbs.eeworld.com.cn/thread-1079275-1-1.html
- https://whycan.com/t_2297.html
- https://www.amobbs.com/thread-5708955-1-1.html#postmessage_10672027

Also http://dengkanwen.com/ seems to run http://www.opensoc.cn/ which sells expansion boards that were mentioned in some other issue, but most of their documentation seems broken.

tpecar, 2021-01-05:
After looking around various forums and looking through the materials that people have gathered on the Telegram group, we can assume that the original schematic file was never shared.

So if we want cross probing, we'll have to recreate it.

I will close this issue when I transfer the useful information to the Wiki.

closed 2021-01-05 by tpecar