xjtuecho/EBAZ4205 #6: Purpose of the telegram channel?

tpecar, 2020-12-20:
I like the idea of having a chat, but I don't understand why it's read-only for everyone?

It would be nice if we can move some the discussion there, so that we don't litter github issues.

Or possibly even consider using a different communication channel
- IRC: register a channel on [freenode.net](https://meta.wikimedia.org/wiki/IRC/Instructions) / [OFTC.net](https://www.oftc.net/Services/) and set up a guard
- Discord
- etc.

xjtuecho, 2020-12-21:
What we need is a telegram group, not channel. I have replaced the telegram channel with a telegram group.
Github discussion is also available.

tpecar, 2020-12-21:
Ok, I wasn't aware of these features, both on github and telegram :sweat_smile:
Thank you for setting this up, this will work.

closed 2020-12-21 by tpecar