xjtuecho/EBAZ4205 #7: Frequency info

doppioandante, 2020-12-20:
If I can correctly interpret the pics, the external crystal is at 25000MHz, is this correct? Does the FPGA run at this frequency?

tpecar, 2020-12-20:
The 25MHz crystal (Y3) is for the ethernet phy.

Zynq uses the 33.333MHz crystal (X8), which can then be internally multiplied by a PLL - this depends on your design, the firmware already present on the flash chip seems to use 666.66 MHz for the ARM core (judging from the BogoMIPS).

The overview datasheet specifies max frequency 667MHz for grade 1 Z-7010  used on these boards. But again, the actual frequency you can use will be determined by your design.

doppioandante, 2020-12-20:
Thank you, that's exactly what I needed to know

closed 2020-12-20 by doppioandante