xjtuecho/EBAZ4205 #9: Mention Wiki + Github discussions in README.md

tpecar, 2021-01-04:
While the discussions feature has been enabled, I guess that people just assume that we don't use it and rather ask in the Telegram group.

Nothing wrong with that, but there are questions that pop up all the time, and some discussions that turn into long sagas which make the rest of the discussion hard to follow.

Linking these resources in the README.md would give the people a bit more confidence to start using them.

tpecar, 2021-01-04:
I would also suggest that you provide additional info on the purpose of each communication channel.

My suggestions would be:

- [Telegram group](https://t.me/ebaz4205) - general chat, quick Q&A
- [Wiki](https://github.com/xjtuecho/EBAZ4205/wiki) - FAQ, tutorials
- [Github discussions](https://github.com/xjtuecho/EBAZ4205/discussions) - troubleshooting help, projects, development

xjtuecho, 2021-01-05:
Done.

tpecar, 2021-01-05:
Thank you, however you pasted the wrong description at Wiki.

tpecar, 2021-01-19:
Fixed the description along with other stuff in #18 .

closed 2021-01-19 by tpecar