xjtuecho/EBAZ4205 #13: Missing Components

firepower9966, 2021-01-06:
I just received my EBAZ4205 from aliexpress, I noticed some components are missing compared to PCB in Photo
Missing: (next to J5, J3)
U67, R2546, R2544, R2545
U68, R2548, R2549, R2547
U69, R2554, R2551
U70, R2557, R2555
Added: 
R2553, R2552, R2554, R2557

Missing: (Lan)
Y3 25MHz xtal
C377, C380 22pF

tpecar, 2021-01-06:
There are at least two known hardware revisions of the board:
1. version, pictured in the repo README, which has the phy + optocouplers
2. version (**what you have**), where these components (+various resistors and caps) are missing

They presumably use the same PCB. It has been speculated that they have different software / bitstream configurations and that in the 2nd version the FPGA generates the clock for the phy. 

However, I haven't looked into this myself, other members may have more information on this.

Kudos for identifying the missing components and measuring their values - I suggest that you add them to [hardware build states wiki page](https://github.com/xjtuecho/EBAZ4205/wiki/Hardware-Build-States.-What-you-may-or-not-have-fitted-on-your-board.).

tpecar, 2021-01-06:
Also, it's probably a good idea that people start nagging their sellers to partially refund them for falsely advertising their product and to force them to update the product image.

Most, if not all, sellers use the 1. version for the product image (which was probably made early in the devices product cycle) but sell you the 2nd. There have been numerous reports about this already.

firepower9966, 2021-01-06:
my seller sold both types, i selected the cheaper version which had no sdcard and jumpers, but they were supplied to be soldered by user. did not realized other components were missing though, pictures of both boards were shown.

nic3-14159, 2021-01-06:
Mine was missing the opto-isolators and 25MHz crystal as well, where the opto-isolators were bypassed by those extra 33 ohm resistors. I took a look at the schematic and my board and found that the 10k ohm resistors R1344 and R1361 are pull-ups to 3.3V on pin 3 (labeled as SPEED on the board) of both J3 and J5. So using logic levels of less than 3.3V on those pins should probably be avoided if using a board with the missing components. Alternatively, just remove those resistors.

Looking at the schematic again, it seems like if the opto-isolators are present, R2350 (the board seems to say R2550, but the schematic says R2350) and R2556 pull up pin 4 on J3/J5 to VCC8, which is whatever voltage is on J4 or pin 2 on J3/J5. Those resistors are missing on the 2nd version of the board. With the opto-isolator, pin 4 (labled as PWM on the board) can only be used as an output, and the pull-up resistors means that the logical high voltage will be whatever VCC8 is. So if you have added the D24 diode to use J3/4/5 as the main power source, make sure that whatever is connected to pin 4 pin J3/5 can handle that voltage if using a board with the isolators.

I'll add this info to the wiki.