xjtuecho/EBAZ4205 #17: Error running bootm command

dave18, 2021-01-18:
Hi

When trying to reset the password I get the following error when running the bootm command 'waiting for root device /dev/mtdblock6', the full Putty log is

------------------------------------------------------------
`
U-Boot 2015.07 (May 18 2017 - 22:22:09 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
NAND:  128 MiB
MMC:   zynq_sdhci: 0
Invalid bus 0 (err=-1)
*** Warning - spi_flash_probe() failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Net:   Gem.e000b000
Hit 'd' to stop autoboot(eric): 0
zynq-uboot> setenv nandboot "echo Copying Linux from NAND flash to RAM... && nand info && run nandroot;nand read 0x100000 0x2220000 0x300000 && fpga loadb 0 0x100000 0x300000 && nand read ${kernel_load_address} 0x300000 ${kernel_size} && nand read ${devicetree_load_address} 0x800000 ${devicetree_size}"
zynq-uboot> run nandboot
Copying Linux from NAND flash to RAM...

Device 0: nand0, sector size 128 KiB
  Page size       2048 b
  OOB size          64 b
  Erase size    131072 b
  subpagesize      512 b
  options     0x       0
  bbt options 0x   20000
## Error: "nandroot" not defined

NAND read: device 0 offset 0x2220000, size 0x300000
 3145728 bytes read: OK
  design filename = "top_ctrl;UserID=0XFFFFFFFF;Version=2016.4"
  part number = "7z010clg400"
  date = "2017/05/29"
  time = "11:17:01"
  bytes in bitstream = 2083740
zynq_align_dma_buffer: Align buffer at 100068 to 100000(swap 1)

NAND read: device 0 offset 0x300000, size 0x500000
 5242880 bytes read: OK

NAND read: device 0 offset 0x800000, size 0x20000
 131072 bytes read: OK
zynq-uboot> setenv bootargs 'console=ttyPS0,115200 root=/dev/mtdblock6 rootfstype=jffs2 noinitrd rw rootwait reboot=cold,hard emergency init=/bin/sh'
zynq-uboot> bootm ${kernel_load_address} - ${devicetree_load_address} init=/bin/sh
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-4.6.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3820688 Bytes = 3.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Device Tree to 0f2fe000, end 0f303a6b ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.6.0-xilinx (root@alvin) (gcc version 5.2.0 (GCC) ) #5 SMP PREEMPT Mon May 8 14:36:49 CST 2017
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: xlnx,zynq-7000
cma: Reserved 16 MiB at 0x0e000000
[... 878 lines cut ...]
CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.6.0-xilinx #5
Hardware name: Xilinx Zynq Platform
[<c010e3e4>] (unwind_backtrace) from [<c010a624>] (show_stack+0x10/0x14)
[<c010a624>] (show_stack) from [<c02cfd98>] (dump_stack+0x80/0xa0)
[<c02cfd98>] (dump_stack) from [<c010cb24>] (ipi_cpu_stop+0x3c/0x70)
[<c010cb24>] (ipi_cpu_stop) from [<c010d39c>] (handle_IPI+0x64/0x84)
[<c010d39c>] (handle_IPI) from [<c01013e4>] (gic_handle_irq+0x74/0x90)
[<c01013e4>] (gic_handle_irq) from [<c010b014>] (__irq_svc+0x54/0x90)
Exception stack(0xc0901f30 to 0xc0901f78)
1f20:                                     00000000 00000001 0f593000 cfdd31c0
1f40: 00000001 cfdd25c0 843407ba 00000001 812def3a 00000001 00000000 00000001
1f60: 00000008 c0901f80 c0491c30 c0491c54 60000013 ffffffff
[<c010b014>] (__irq_svc) from [<c0491c54>] (cpuidle_enter_state+0xe0/0x1b4)
[<c0491c54>] (cpuidle_enter_state) from [<c014a0dc>] (cpu_startup_entry+0x184/0x1d0)
[<c014a0dc>] (cpu_startup_entry) from [<c0800bb4>] (start_kernel+0x2f0/0x350)
[<c0800bb4>] (start_kernel) from [<0000807c>] (0x807c)
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,5)
`
-----------------------------------------------------------------------------------------------

Grateful for any help on how to get this working

Thanks

Dave

dave18, 2021-01-18:
I've also tried mtdblock4 as this seems to have the jffs2 file system on it, this give the following

U-Boot 2015.07 (May 18 2017 - 22:22:09 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
NAND:  128 MiB
MMC:   zynq_sdhci: 0
Invalid bus 0 (err=-1)
*** Warning - spi_flash_probe() failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Net:   Gem.e000b000
Hit 'd' to stop autoboot(eric): 0
zynq-uboot> setenv nandboot "echo Copying Linux from NAND flash to RAM... && nand info && run nandroot;nand read 0x100000 0x2220000 0x300000 && fpga loadb 0 0x100000 0x300000 && nand read ${kernel_load_address} 0x300000 ${kernel_size} && nand read ${devicetree_load_address} 0x800000 ${devicetree_size}"
zynq-uboot> run nandboot
Copying Linux from NAND flash to RAM...

Device 0: nand0, sector size 128 KiB
  Page size       2048 b
  OOB size          64 b
  Erase size    131072 b
  subpagesize      512 b
  options     0x       0
  bbt options 0x   20000
## Error: "nandroot" not defined

NAND read: device 0 offset 0x2220000, size 0x300000
 3145728 bytes read: OK
  design filename = "top_ctrl;UserID=0XFFFFFFFF;Version=2016.4"
  part number = "7z010clg400"
  date = "2017/05/29"
  time = "11:17:01"
  bytes in bitstream = 2083740
zynq_align_dma_buffer: Align buffer at 100068 to 100000(swap 1)

NAND read: device 0 offset 0x300000, size 0x500000
 5242880 bytes read: OK

NAND read: device 0 offset 0x800000, size 0x20000
 131072 bytes read: OK
zynq-uboot> setenv bootargs 'console=ttyPS0,115200 root=/dev/mtdblock4 rootfstype=jffs2 noinitrd rw rootwait reboot=cold,hard emergency init=/bin/sh'
zynq-uboot> bootm ${kernel_load_address} - ${devicetree_load_address} init=/bin/sh
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-4.6.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3820688 Bytes = 3.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Device Tree to 0f2fe000, end 0f303a6b ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.6.0-xilinx (root@alvin) (gcc version 5.2.0 (GCC) ) #5 SMP PREEMPT Mon May 8 14:36:49 CST 2017
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: xlnx,zynq-7000
cma: Reserved 16 MiB at 0x0e000000
Memory policy: Data cache writealloc
percpu: Embedded 12 pages/cpu @cfdcf000 s19776 r8192 d21184 u49152
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
Kernel command line: console=ttyPS0,115200 root=/dev/mtdblock4 rootfstype=jffs2 noinitrd rw rootwait reboot=cold,hard emergency init=/bin/sh
[... 155 lines cut ...]
ALSA device list:
  No soundcards found.
jffs2: Empty flash at 0x009030f0 ends at 0x00903800
jffs2: Empty flash at 0x009048f0 ends at 0x00905000
VFS: Mounted root (jffs2 filesystem) on device 31:4.
devtmpfs: error mounting -2
Freeing unused kernel memory: 1024K (c0800000 - c0900000)
Kernel panic - not syncing: Requested init /bin/sh failed (error -2).
CPU0: stopping
CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.6.0-xilinx #5
Hardware name: Xilinx Zynq Platform
[<c010e3e4>] (unwind_backtrace) from [<c010a624>] (show_stack+0x10/0x14)
[<c010a624>] (show_stack) from [<c02cfd98>] (dump_stack+0x80/0xa0)
[<c02cfd98>] (dump_stack) from [<c010cb24>] (ipi_cpu_stop+0x3c/0x70)
[<c010cb24>] (ipi_cpu_stop) from [<c010d39c>] (handle_IPI+0x64/0x84)
[<c010d39c>] (handle_IPI) from [<c01013e4>] (gic_handle_irq+0x74/0x90)
[<c01013e4>] (gic_handle_irq) from [<c010b014>] (__irq_svc+0x54/0x90)
Exception stack(0xc0901f30 to 0xc0901f78)
1f20:                                     00000000 00000000 0f593000 cfdd31c0
1f40: 00000000 cfdd25c0 51c108d0 00000000 5163d45f 00000000 00000000 00000000
1f60: 00000008 c0901f80 c0491c30 c0491c54 60000013 ffffffff
[<c010b014>] (__irq_svc) from [<c0491c54>] (cpuidle_enter_state+0xe0/0x1b4)
[<c0491c54>] (cpuidle_enter_state) from [<c014a0dc>] (cpu_startup_entry+0x184/0x1d0)
[<c014a0dc>] (cpu_startup_entry) from [<c0800bb4>] (start_kernel+0x2f0/0x350)
---[ end Kernel panic - not syncing: Requested init /bin/sh failed (error -2).

DavidJRichards, 2021-01-18:
Greetings dave18, I had a problem using the commands given here but found a way to make it work for me. The instructions here were supposed to have been fixed but perhaps your system is different or maybe they haven't been tested. The commands which actually worked for me are shown here:  https://github.com/DavidJRichards/EBAZ4205/tree/master/uboot There is also some background information which may enable you to craft a working set of commands. hth David.

dave18, 2021-01-18:
Hi David, thanks for your response.

Hmm, your commands are identical to the last 2 commands given here, which suggest they have been fixed.  On your site there is no command to create the boot image in memory, so does your board automatically create an image?

When I run it I get 'ERROR: can't get kernel image!'

nic3-14159, 2021-01-18:
There might be another reason why your board isn't working. To me it looks like Linux is only detecting 6 mtd partitions. If you look [here](https://github.com/xjtuecho/EBAZ4205/blob/master/proc/mtd), you can see that there should be 9 mtd partitions. The partition mtd6 should be after the "nand-bitstream" partition. Maybe there's an issue with the NAND flash on your board that is causing those partitions not to be detected? I haven't figured out how to list partitions in uboot though.

DavidJRichards, 2021-01-18:
HI Dave, sorry, Having a bad day, I was looking in the wrong place.  does your board boot to a Linux command prompt without any intervention? how are the boot options set? Somebody here has a board with duff one or more nand blocks. you can follow what is supposed to happen if you issue a 'printenv' command before letting the system boot. Linux kernel memory is filled from nand with the commands -- see below  David.

'''
--- original nandboot: loads fpga, no ramdisk
nandboot=echo Copying Linux from NAND flash to RAM... && 
nand info && 
run nandroot;
nand read 0x100000 0x2220000 0x300000 && 
fpga loadb 0 0x100000 0x300000 && 
nand read ${kernel_load_address} 0x300000 ${kernel_size} && 
nand read ${devicetree_load_address} 0x800000 ${devicetree_size} && 
bootm ${kernel_load_address} - ${devicetree_load_address}

--- helper
nandroot=setenv bootargs 'console=ttyPS0,115200 root=/dev/mtdblock6 rootfstype=jffs2 noinitrd rw rootwait'
'''

dave18, 2021-01-18:
I notice that when I call 'run nandboot' it gives the following error when it tries to run nandroot '## Error: "nandroot" not defined'.  printenv does not have seem to have an environment for nandroot anywhere, I've included the output at the bottom of this post.

Also, if my nand is corrupted is there any way to flash a correct image?

If I just let it boot it does seem to start successfully (until it can't reach an IP address)

printenv output

zynq-uboot> printenv
baudrate=115200
bitstream_image=system.bit.bin
boot_image=BOOT.bin
boot_size=0xF00000
bootcmd=run $modeboot
bootdelay=3
bootenv=uEnv.txt
devicetree_image=devicetree.dtb
devicetree_load_address=0x2000000
devicetree_size=0x20000
dfu_mmc=run dfu_mmc_info && dfu 0 mmc 0
dfu_mmc_info=set dfu_alt_info ${kernel_image} fat 0 1\\;${devicetree_image} fat 0 1\\;${ramdisk_image} fat 0 1
dfu_ram=run dfu_ram_info && dfu 0 ram 0
dfu_ram_info=set dfu_alt_info ${kernel_image} ram 0x3000000 0x500000\\;${devicetree_image} ram 0x2A00000 0x20000\\;${ramdisk_image} ram 0x2000000 0x600000
ethact=Gem.e000b000
ethaddr=00:0a:35:00:01:22
fdt_high=0x20000000
gatewayip=192.168.0.1
importbootenv=echo Importing environment from SD ...; env import -t ${loadbootenv_addr} $filesize
initrd_high=0x20000000
ipaddr=192.168.0.115
jtagboot=echo TFTPing Linux to RAM... && tftpboot ${kernel_load_address} ${kernel_image} && tftpboot ${devicetree_load_address} ${devicetree_image} && tftpboot ${ramdisk_load_address} ${ramdisk_image} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
kernel_image=uImage
kernel_load_address=0x2080000
kernel_size=0x500000
loadbit_addr=0x100000
loadbootenv=load mmc 0 ${loadbootenv_addr} ${bootenv}
loadbootenv_addr=0x2000000
mmc_loadbit=echo Loading bitstream from SD/MMC/eMMC to RAM.. && mmcinfo && load mmc 0 ${loadbit_addr} ${bitstream_image} && fpga loadb 0 ${loadbit_addr} ${filesize}
modeboot=nandboot
nandboot=echo Copying Linux from NAND flash to RAM... && nand read 0x100000 0x2220000 0x300000 && fpga loadb 0 0x100000 0x300000 && nand read ${kernel_load_address} 0x300000 ${kernel_size} && nand read ${devicetree_load_address} 0x800000 ${devicetree_size} && echo Copying ramdisk... && nand read ${ramdisk_load_address} 0x820000 ${ramdisk_size} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
nandboot0=echo Copying Linux from NAND flash to RAM... && nand read ${kernel_load_address} 0x300000 ${kernel_size} && nand read ${devicetree_load_address} 0x800000 ${devicetree_size} && echo Copying ramdisk... && nand read ${ramdisk_load_address} 0x820000 ${ramdisk_size} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
norboot=echo Copying Linux from NOR flash to RAM... && cp.b 0xE2100000 ${kernel_load_address} ${kernel_size} && cp.b 0xE2600000 ${devicetree_load_address} ${devicetree_size} && echo Copying ramdisk... && cp.b 0xE2620000 ${ramdisk_load_address} ${ramdisk_size} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
preboot=if test $modeboot = sdboot && env run sd_uEnvtxt_existence_test; then if env run loadbootenv; then env run importbootenv; fi; fi;
qspiboot=echo Copying Linux from QSPI flash to RAM... && sf probe 0 0 0 && sf read ${kernel_load_address} 0x100000 ${kernel_size} && sf read ${devicetree_load_address} 0x600000 ${devicetree_size} && echo Copying ramdisk... && sf read ${ramdisk_load_address} 0x620000 ${ramdisk_size} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
ramdisk_image=uramdisk.image.gz
ramdisk_load_address=0x4000000
ramdisk_size=0xA00000
rsa_jtagboot=echo TFTPing Image to RAM... && tftpboot 0x100000 ${boot_image} && zynqrsa 0x100000 && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
rsa_nandboot=echo Copying Image from NAND flash to RAM... && nand read 0x100000 0x0 ${boot_size} && zynqrsa 0x100000 && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
rsa_norboot=echo Copying Image from NOR flash to RAM... && cp.b 0xE2100000 0x100000 ${boot_size} && zynqrsa 0x100000 && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
rsa_qspiboot=echo Copying Image from QSPI flash to RAM... && sf probe 0 0 0 && sf read 0x100000 0x0 ${boot_size} && zynqrsa 0x100000 && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
rsa_sdboot=echo Copying Image from SD to RAM... && load mmc 0 0x100000 ${boot_image} && zynqrsa 0x100000 && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}
sd_uEnvtxt_existence_test=test -e mmc 0 /uEnv.txt
sdboot=if mmcinfo; then run uenvboot; echo Copying Linux from SD to RAM... && load mmc 0 ${kernel_load_address} ${kernel_image} && load mmc 0 ${devicetree_load_address} ${devicetree_image} && load mmc 0 ${ramdisk_load_address} ${ramdisk_image} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}; fi
serverip=192.168.0.104
stderr=serial
stdin=serial
stdout=serial
thor_mmc=run dfu_mmc_info && thordown 0 mmc 0
thor_ram=run dfu_ram_info && thordown 0 ram 0
uenvboot=if run loadbootenv; then echo Loaded environment from ${bootenv}; run importbootenv; fi; if test -n $uenvcmd; then echo Running uenvcmd ...; run uenvcmd; fi
usbboot=if usb start; then run uenvboot; echo Copying Linux from USB to RAM... && load usb 0 ${kernel_load_address} ${kernel_image} && load usb 0 ${devicetree_load_address} ${devicetree_image} && load usb 0 ${ramdisk_load_address} ${ramdisk_image} && bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}; fi

Environment size: 5290/131068 bytes

dave18, 2021-01-18:
Ok, I've soldered on a TF Card slot and can access it from uboot.  Would someone be able to provide me with a good flash image, then I can assume I could try:

fatload mmc 0 0x100000 image.bin
nand scrub -y 0x0 0x8000000            (mine is currently reporting bad blocks)
nand write 0x100000 0x0 0x8000000

and that should restore the nand?

DavidJRichards, 2021-01-18:
I dumped the nand images to file when I got my board, haven't dared tried writing to nand yet though and cant vouch for their contents although they look plausible. images here: https://github.com/DavidJRichards/EBAZ4205/tree/master/nand I dont know if the uboot  nand writing functions work either.

dave18, 2021-01-18:
Thanks David

Well, it seems to have worked, took a couple of goes and I ended up doing a full erase and then writing all the MDTs back again and it is fine other than a few jffs CRC complaints.

I take it you had already renamed the bitminer script before you dumped the nand as it was already renamed for me.

Thanks again

Dave

DavidJRichards, 2021-01-18:
Thanks for the good news Dave, could you post a transcript please. Yes I guess I had already reamed the script, it was the first thing I did, Kind regards, David.

dave18, 2021-01-18:
U-Boot 2015.07 (May 18 2017 - 22:22:09 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
NAND:  128 MiB
MMC:   zynq_sdhci: 0
Invalid bus 0 (err=-1)
*** Warning - spi_flash_probe() failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Net:   Gem.e000b000
Hit 'd' to stop autoboot(eric): 0
zynq-uboot> ls mmc 0
            system volume information/
 16777216   mtd4ro.bin
  8388608   mtd5ro.bin
 67108864   mtd6ro.bin
 20840448   mtd7ro.bin
  2097152   mtd8ro.bin
  3145728   mtd0ro.bin
  5242880   mtd1ro.bin
   131072   mtd2ro.bin
 10485760   mtd3ro.bin

9 file(s), 1 dir(s)

zynq-uboot> load mmc 0 0x100000 mtd8ro.bin
reading mtd8ro.bin
2097152 bytes read in 189 ms (10.6 MiB/s)

zynq-uboot> nand write 0x100000 0x7e00000 0x200000

NAND write: device 0 offset 0x7e00000, size 0x200000
Attempt to write outside the flash area
 0 bytes written: ERROR

**** This is because I have bad blocks at the end of my NAND

zynq-uboot> load mmc 0 0x100000 mtd7ro.bin
reading mtd7ro.bin
20840448 bytes read in 1743 ms (11.4 MiB/s)
zynq-uboot> nand write 0x100000 0x6a20000 0x13e0000

NAND write: device 0 offset 0x6a20000, size 0x13e0000
 20840448 bytes written: OK
zynq-uboot> nand bad

Device 0 bad blocks:
  07f60000
  07f80000
  07fa0000
  07fc0000
  07fe0000
zynq-uboot> nand scrub -y 0x7e00000 0x200000

NAND scrub: device 0 offset 0x7e00000, size 0x200000
Erasing at 0x7fe0000 -- 100% complete.
OK
zynq-uboot> nand bad

Device 0 bad blocks:
  07f80000
  07fa0000
  07fc0000
  07fe0000
zynq-uboot> nand scrub -y 0x7f80000 0x20000

[... 1020 lines cut ...]
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
random: dd urandom read with 2 bits of entropy available
Sat Jan 19 21:08:14 UTC 2019
Starting internet superserver: inetd.
INIT: Entering runlevel: 5
Configuring network interfaces... IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
udhcpc (v1.24.1) started
Sending discover...
Sending discover...
Sending discover...
No lease, forking to background
done.
Starting Dropbear SSH server: dropbear.
Starting syslogd/klogd: done

PetaLinux 2016.4 zedboard-zynq7 /dev/ttyPS0

zedboard-zynq7 login: jffs2: notice: (742) check_node_data: wrong data CRC in data node at 0x02539bb8: read 0xc87e348e, calculated 0x67f3ed0a.
root
Password:
root@zedboard-zynq7:~# ls
dwang_btc_miner
root@zedboard-zynq7:~# jffs2: notice: (742) check_node_data: wrong data CRC in data node at 0x019d2c54: read 0x59459af6, calculated 0xb967dedf.
ls
dwang_btc_miner

closed 2021-01-19 by dave18