xjtuecho/EBAZ4205 #28: Weird characters in UART

Guillermo-Ramirez-Jimenez, 2021-02-09:
Hi,
recently, I got one of these boards and I tried to have a look at the SSH in UART. Unfortunately, I get some mixed right and wrong characters.

I tried 2 different USB to TTL converters and cables, but the issue is still there. 115200 bauds.

Am I missing any particular option for the serial connection?

Here you have a sample of the output:

```
.c⸮⸮⸮f⸮⸮⸮⸮⸮"Mcs'dv⸮/2fz⸮ֱ⸮⸮"⸮O3⸮⸮/_⸮^g⸮⸮⸮do&⸮i⸮ɨ⸮/wv⸮⸮⸮w⸮⸮f⸮~⸮⸮⸮⸮-1)
*** Warning - spi_flash_pr⸮be(+$ocZ_6⸮"⸮smgd⸮g⸮|7~⸮z⸮⸮o{vs⸮zl⸮g⸮⸮n⸮⸮⸮⸮?⸮w⸮⸮ ⸮lv⸮⸮2⸮l6w⸮n⸮nz]⸮"⸮⸮v/⸮⸮⸮чcu_⸮⸮⸮;⸮g~f⸮⸮⸮⸮h⸮⸮她⸮⸮Linux from NAND flash to⸮⸮⸮⸮⸮٭⸮7*D⸮vz⸮6⸮⸮⸮d⸮o⸮⸮c٧sۿ6?⸮ƃz⸮bm36⸮⸮⸮⸮⸮b⸮⸮⸮⸮߻睿⸮w⸮⸮2⸮n6⸮⸮2fc'v{⸮խ⸮nsL$V⸮⸮v⸮f⸮~~⸮⸮⸮⸮⸮⸮j

NAND read: device 0 offset 0x2⸮&f¢f⸮⸮z[lf32⸮f⸮v⸮⸮⸮⸮⸮>⸮⸮⸮yg⸮l;⸮⸮⸮{⸮n⸮⸮⸮⸮⸮⸮⸮⸮z{v⸮⸮{t⸮⸮W⸮Vv⸮⸮⸮e⸮⸮~f⸮⸮⸮&⸮00000, size 0x500000
"w'⸮⸮⸮#b⸮⸮#v⸮c⸮7⸮⸮x⸮⸮ 131072 bytes read: OK
## Bootkng kernel⸮⸮⸮⸮wgg⸮s^l⸮o⸮⸮#⸮&⸮&f26⸮f⸮>.
*⸮Lr⸮⸮⸮⸮⸮>c⸮t⸮]⸮Ց⸮?nnf⸮7⸮⸮d⸮~{gvf⸮>⸮L3⸮"{⸮w⸮o⸮⸮f⸮⸮⸮⸮⸮⸮⸮⸮j
   Entry Point:  00008000
 ` veri⸮^⸮⸮9⸮⸮⸮⸮⸮o⸮r⸮⸮O⸮⸮3d⸮n{w˝⸮"⸮wkg⸮wn⸮f⸮⸮⸮"e⸮ ⸮7⸮vw⸮⸮⸮⸮~&n?.⸮   Loading Device ⸮ree to ⸮⸮v⸮⸮⸮l;;⸮⸮ ⸮⸮w⸮6⸮2d⸮⸮;z/nc⸮wcv⸮k⸮~⸮⸮⸮⸮⸮⸮ߥs⸮O>aU&⸮own⸮ou⸮dwo⸮⸮nj⸮⸮⸮{m⸮^;",⸮⸮⸮Fg_⸮g⸮⸮⸮	J⸮⸮5A⸮PREEMPT Mon May 8 14:36:49(Cs⸮f_⸮n-w⸮⸮s⸮w⸮⸮n⸮sg⸮⸮k⸮_⸮3⸮o⸮w⸮⸮|⸮Vo⸮⸮>f⸮V77*⸮{g⸮^⸮/~⸮svw⸮{o0gx,⸮⸮⸮?㟷⸮⸮⸮⸮~⸮⸮⸮⸮⸮Y⸮⸮ٕ⸮⸮16 MiB at 0x0e000000
Memor^H⸮⸮⸮Q,⸮ /⸮⸮⸮ ͽ⸮⸮g⸮⸮⸮?.⸮ۿsw⸮LvKg⸮⸮⸮?⸮⸮⸮鿽g⸮v⸮⸮r⸮l⸮k⸮~w⸮w⸮⸮⸮⸮n6{⸮⸮ ⸮}⸮⸮⸮⸮⸮⸮⸮~⸮⸮⸮⸮⸮ne: console=ttyPS0,115200 root=/dev/mtdbl⸮so⸮$⸮⸮⸮st~⸮ۛ⸮s⸮⸮⸮⸮~⸮⸮.v⸮݃roov⸮o⸮⸮⸮w⸮?+⸮⸮⸮me⸮⸮⸮so{.⸮32⸮⸮⸮;o⸮ֶ;$oOr⸮⸮ƻ⸮⸮⸮⸮⸮⸮⸮vn⸮⸮⸮⸮v⸮⸮⸮⸮⸮⸮⸮ 16384 (order: 4, 65536 bytes)
sY[K⸮⸮⸮2.⸮⸮wZ⸮⸮6⸮n3[fz߷⸮g⸮6O⸮⸮⸮6⸮z⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M⸮⸮⸮⸮Ooֽ⸮⸮r۾⸮⸮7⸮3⸮o2⸮~6⸮gm⸮/n⸮k⸮⸮]X_⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ : 0xffff0000 - 0xffff1000   (⸮"`4"oB;
⸮H2
2⸮zmznvo&⸮zfcv⸮fw"m|f⸮⸮f⸮Lw⸮"⸮f6⸮⸮⸮^'vӍ⸮0v⸮&s3o⸮f⸮2v⸮&72f⸮$eo⸮⸮⸮O⸮N⸮f⸮⸮⸮n ⸮⸮f⸮⸮⸮⸮⸮⸮⸮&B⸮j	⸮
    modules : ⸮⸮⸮⸮0000p0/`0z⸮⸮6⸮⸮⸮"⸮NgV⸮⸮⸮o⸮nr⸮f+2r⸮⸮⸮١; ⸮⸮~7w3⸮\VB⸮⸮2t⸮{fjv⸮#⸮|&⸮⸮⸮⸮c`⸮d7cp⸮24⸮⸮W[Ff⸮;⸮⸮⸮⸮ff~f⸮⸮⸮⸮,'⸮⸮939e60 - 0xc0972764   ( 227 kF)
⸮⸮{_nyf⸮]vo⸮⸮sv⸮⸮{o⸮⸮⸮Gw⸮on{_⸮ϝ⸮⸮⸮⸮⸮⸮⸮>⸮Ko{L⸮⸮7sWs_m6$⸮⸮⸮⸮w⸮f⸮v5.v⸮⸮⸮Ad⸮⸮fV⸮⸮⸮⸮⸮⸮⸮32, nr_cpu_ids=2
NR_IRQS:16 nr_irqs:q7 5>N{V⸮gdnwg⸮⸮⸮;"⸮⸮6⸮Lf:sv⸮+⸮r⸮2⸮3⸮7⸮2⸮^⸮⸮⸮⸮擲w;N⸮*⸮vto⸮⸮⸮fO⸮⸮⸮o⸮/v⸮݃cuy`g⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮j
L2C-310 erratum 769419 Y+⸮⸮u
⸮⸮⸮k⸮⸮⸮g⸮~6?nV;⸮⸮;⸮f⸮w73⸮~⸮⸮v⸮⸮⸮w⸮⸮⸮⸮⸮&Ygf⸮wv⸮"of⸮vdNf;~e⸮v72gO⸮3"⸮n⸮{⸮⸮gn⸮⸮⸮⸮⸮⸮߷⸮*⸮⸮⸮⸮⸮⸮5
L2C-310 cache control⸮ev⸮en⸮bn⸮Y⸮dW⸮⸮>dl7o⸮7;N⸮⸮⸮⸮⸮⸮3gr>⸮⸮⸮⸮t~⸮⸮3f⸮⸮⸮⸮⸮~u⸮,⸮o⸮⸮⸮n{_⸮jg⸮_⸮5⸮⸮so⸮n3⸮fZ].-⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮398046511103ns
clocksource: arm_gloramt⸮oe⸮:d⸮⸮of⸮x⸮⸮⸮f⸮⸮⸮f⸮⸮⸮oe⸮ogl⸮⸮g⸮|v6o⸮~op,⸮⸮⸮l⸮⸮ϑ⸮~wg⸮vso⸮VKf⸮⸮⸮7⸮g⸮⸮w⸮g⸮⸮⸮⸮ycles: 0xffff, max_idle_ns: 537538<77 nsc3w⸮⸮⸮⸮#⸮Lv2f⸮Ns⸮⸮⸮?&z⸮w⸮7s⸮⸮⸮ݿ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮}6⸮ߥ~g⸮vm⸮W⸮⸮ws_⸮⸮>⸮n⸮r⸮⸮"F⸮g⸮⸮⸮e⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮'⸮⸮⸮⸮⸮j⸮⸮⸮⸮յ⸮ 301
Mount-cacje jasj`⸮,6⸮⸮⸮~t⸮~⸮>'v⸮⸮⸮3jK⸮⸮r~~f;⸮⸮7$⸮/wݽw⸮⸮⸮⸮~⸮⸮⸮g⸮⸮mɭ⸮⸮&⸮⸮⸮⸮v⸮⸮⸮v{⸮⸮vdo=⸮(wU⸮Fﾕ⸮⸮⸮⸮o⸮⸮󂂂⸮⸮⸮⸮j
Setting up static identity map`⸮ov⸮0⸮Lf3⸮⸮⸮?⸮n⸮Lf⸮>⸮.Uw⸮⸮w⸮⸮V6 ⸮6⸮⸮⸮⸮⸮⸮⸮⸮swOݥsg⸮ow⸮⸮⸮"⸮~⸮⸮⸮⸮2⸮⸮7{t⸮N⸮Lt⸮w⸮⸮v⸮fvW⸮⸮~⸮⸮⸮⸮⸮ѵ⸮⸮⸮⸮ initialized
VFP support ⸮Ly⸮N⸮ѳս⸮⸮vob⸮lvoc⸮~wot⸮6o⸮yv⸮⸮fbv^⸮⸮⸮⸮f⸮⸮⸮c⸮⸮⸮⸮g⸮o⸮⸮3⸮Nf⸮2⸮O⸮⸮⸮3W⸮⸮⸮nw*⸮?⸮⸮⸮⸮⸮g⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ѕ⸮5
NET: Registered protocon`fcminy⸮LsHD7C~⸮nwo⸮⸮cxWv0⸮⸮.⸮⸮so⸮⸮⸮r\⸮⸮⸮⸮⸮⸮⸮⸮ݻf&_n{/3⸮⸮w[>w⸮/⸮>w/o⸮⸮⸮⸮d{⸮⸮Ł⸮|⸮⸮⸮⸮~]LW⸮⸮⸮⸮⸮⸮H[X⸮⸮յ⸮watchpoint size is 4 bytes.
*z⸮nq⸮og⸮ f⸮2⸮>f⸮.o⸮gn⸮lw2⸮ol; ⸮Ow⸮⸮⸮⸮⸮⸮wvn⸮O"⸮e⸮vnv=⸮i~⸮⸮o⸮⸮ѝ⸮#k⸮~wo⸮⸮ze⸮⸮⸮wf_⸮⸮~⸮⸮⸮⸮ٕɁusbfs
usbcore: registered new z⸮⸮ə⸮⸮⸮⸮⸮rz׶⸮n⸮⸮⸮w⸮⸮⸮~⸮⸮⸮ow⸮]⸮⸮⸮⸮⸮⸮o⸮⸮re1k⸮⸮gv⸮s⸮~⸮2⸮⸮⸮x$⸮⸮⸮s⸮⸮$;⸮⸮r⸮*V^m⸮w⸮⸮vwv⸮w⸮w⸮⸮⸮ftware ver. 5.3.6 - Copyright 2405-⸮04⸮ v⸮⸮⸮fo⸮T⸮⸮⸮⸮wi^>⸮o{}wo⸮z⸮⸮?o⸮jV⸮⸮⸮⸮⸮⸮g⸮v_⸮gg⸮_⸮O3⸮⸮v{>v⸮o⸮~n+z⸮_⸮J6⸮⸮⸮ͽ⸮͍⸮⸮d⸮t?⸮⸮⸮⸮⸮⸮⸮E⸮⸮⸮⸮5
NET: Registered protocol gamily!2S⸮⸮T.⸮ve⸮~o⸮z⸮$;s⸮nw⸮⸮⸮w⸮⸮⸮⸮⸮⸮6⸮⸮f⸮⸮⸮⸮⸮l⸮f⸮2{v{⸮⸮lf?r~~⸮⸮zf⸮|6⸮݇so⸮V⸮V⸮wρ~⸮w⸮f_⸮⸮⸮⸮g⸮⸮⸮͡⸮table entries: 256 (order: L	Ɗʖ⸮⸮ѝ⸮⸮⸮w6⸮r⸮⸮6g{$⸮~.⸮d[t⸮~⸮⸮⸮?>⸮⸮⸮⸮g⸮ρ⸮c[{v⸮⸮:7⸮⸮so{7vr⸮o⸮⸮rvJ⸮⸮wn{⸮zdE⸮⸮⸮⸮⸮⸮o⸮⸮⸮⸮⸮⸮^H5⸮ Registered tcp transport mo⸮ulm.
⸮*⸮('⸮⸮⸮voٷ⸮g⸮l⸮o⸮⸮w⸮⸮;"fz⸮⸮g⸮⸮fv⸮n⸮⸮⸮֦3o⸮׽t{⸮nvv⸮z⸮⸮n⸮cgvt⸮~2+s⸮LW⸮⸮O⸮⸮}v⸮⸮⸮⸮~|⸮⸮⸮⸮⸮X⸮ɭ⸮⸮⸮͕⸮⸮ timestamp_bits=28 mcx_ordgvOf7Mկ{e⸮⸮⸮fe⸮⸮w:⸮⸮⸮⸮:f⸮⸮⸮{o⸮L⸮M⸮⸮:⸮y⸮⸮⸮⸮s⸮⸮⸮⸮⸮g⸮_⸮o⸮⸮]v_⸮⸮ՙ⸮wgz.wo⸮⸮u
Z⸮⸮~⸮⸮w_}w⸮f~⸮⸮⸮[[XK⸮⸮30 f8003000.dmac: Loaded drives f⸮rd⸮nw'f⸮;o?⸮O⸮⸮2s1w;/⸮{⸮⸮&⸮Nfד⸮⸮⸮o⸮⸮⸮⸮⸮⸮{e⸮⸮⸮⸮w2~~7⸮⸮'_fNo⸮⸮⸮22^h;oՃ=$L⸮3M⸮⸮wgeg_⸮v~⸮⸮⸮⸮⸮⸮⸮⸮⸮r⸮⸮ɥ⸮⸮⸮ ttyPS1 at MMIO 0xe00800⸮Ld)⸮⸮⸮?⸮L3⸮&⸮|w⸮e⸮⸮⸮?f⸮⸮⸮⸮'⸮k⸮Xn⸮~w⸮v⸮2⸮⸮6⸮NgWMn⸮2Zn⸮|nsm⸮⸮⸮so⸮g⸮⸮pS1d⸮~⸮|⸮⸮⸮⸮⸮⸮⸮⸮⸮,֖J⸮ɕ⸮⸮⸮⸮0xf8007000 to d0876004⸮[d⸮_⸮o⸮⸮~m⸮v⸮⸮⸮07⸮f;0v~&7º⸮{⸮⸮⸮⸮⸮o⸮⸮⸮⸮⸮o⸮gv⸮⸮e⸮|;⸮⸮⸮⸮⸮?⸮⸮pmf_⸮6g⸮⸮w{7⸮⸮⸮⸮⸮/v:s'M⸮l~⸮⸮⸮o⸮⸮⸮C⸮}⸮⸮Y⸮⸮⸮00.ethernet eth0: Cadence GEM ⸮Y⸮⸮⸮0⸮L⸮:${_v⸮g⸮f⸮O⸮⸮"o⸮~v⸮g⸮0⸮'⸮O⸮⸮⸮⸮⸮⸮⸮⸮>⸮⸮⸮ ⸮⸮&w⸮⸮?dj⸮⸮_⸮n⸮z^5⸮⸮r⸮yn⸮⸮⸮⸮n⸮t⸮߻⸮⸮⸮⸮⸮⸮⸮⸮I⸮⸮I=⸮1000 Network Driver -(3.⸮⸮n{
⸮]fS⸮.⸮ ⸮⸮⸮⸮ot⸮⸮⸮?⸮⸮9&⸮Nfq7⸮v⸮W⸮⸮⸮⸮⸮w⸮vnfrm⸮⸮wNt⸮⸮⸮vK"n⸮⸮
⸮;vWg⸮⸮۵⸮⸮⸮m⸮⸮wof⸮⸮g~⸮⸮⸮⸮⸮⸮⸮driver usb-storage
m⸮ͷ⸮⸮⸮$⸮[2o⸮ݷ3f{o⸮>⸮ｷn⸮⸮["lo⸮⸮⸮⸮⸮o_d⸮J⸮Wٷ⸮2⸮}⸮w⸮n⸮⸮⸮⸮u>r⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮563_write_block_data: evr=-91$x]⸮?6{Od/ٷ?4n⸮⸮vO⸮⸮;7⸮&v?6⸮MM/⸮=⸮⸮⸮⸮⸮⸮u=⸮⸮⸮k⸮?⸮/⸮⸮⸮0⸮nj⸮s⸮n310⸮⸮omo$e⸮⸮ކ⸮⸮⸮⸮⸮~EDAC MC: ECC not enabled-JXZ[m⸮⸮^⸮W:⸮⸮O⸮6⸮g⸮߶⸮w⸮~nw⸮⸮sd~⸮⸮2o⸮⸮r⸮]⸮nͷ⸮nS⸮nOgZo7⸮f['⸮⸮⸮g&⸮,⸮⸮[⸮dZ⸮⸮F⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮?⸮E!
%⸮controller on e0100000.sdxci0[YL&S⸮⸮⸮.w⸮jg⸮⸮}⸮⸮g⸮w⸮⸮⸮⸮⸮{⸮ϕ⸮ag⸮ow⸮⸮⸮onn(w7=ou.⸮⸮K⸮⸮ registered new interface drm⸮ev⸮]n<⸮⸮u.>6/⸮⸮ug⸮⸮⸮nq~f⸮⸮⸮⸮⸮,W⸮⸮ɁID: 0xef, Chip ID: 0xf1
⸮~⸮2⸮⸮o⸮w⸮n⸮⸮~⸮N)⸮⸮⸮f⸮⸮3⸮ ⸮'⸮⸮k⸮g⸮⸮nofX]⸮⸮⸮⸮~⸮⸮472, version 0x01
fZ}⸮ｿ⸮d⸮~v⸮$⸮⸮ff^⸮v⸮⸮Ù⸮L⸮⸮f'gY⸮⸮⸮⸮f{{⸮⸮ooO⸮ ⸮~fW⸮⸮⸮6⸮Lw⸮⸮g⸮⸮⸮⸮⸮00007fa0000
nand_read_bbt: bad`bno⸮kh|]v⸮⸮撢⸮7f<Lw⸮~/;⸮⸮ӧs⸮~⸮⸮wm⸮^⸮o⸮⸮u⸮⸮⸮⸮~M⸮fǊ⸮⸮⸮6⸮Lw⸮{&⸮⸮2⸮l⸮⸮t⸮LwK6⸮⸮⸮ѵ2⸮b⸮⸮⸮_⸮p~L⸮򂂂j⸮⸮⸮00000800000 : "nand-linux"
t⸮2⸮NfNgW⸮⸮⸮/⸮~7w⸮⸮⸮:⸮NfW⸮⸮⸮⸮v>⸮⸮⸮⸮s&c⸮p⸮2⸮⸮⸮⸮⸮000002220000 : "nand-jffs2"
⸮~⸮Ls2v⸮&w2v{lo0v⸮N⸮ײ⸮⸮⸮⸮gd⸮⸮~⸮Lf⸮6e30⸮l&⸮⸮v⸮c⸮⸮⸮nvo⸮⸮fG⸮⸮?fw⸮⸮⸮g4⸮lw⸮⸮f#nz⸮⸮rg⸮_⸮ՙw;t⸮⸮⸮nsw⸮⸮⸮yw⸮⸮⸮w⸮ve⸮⸮
gѧt⸮⸮⸮⸮k⸮o&⸮kv{wz⸮NET: Registmred pzotoco⸮HvV⸮⸮⸮^⸮X⸮⸮c{#⸮⸮⸮⸮^G⸮'⸮W⸮⸮⸮v{⸮6⸮⸮vko⸮&v⸮o⸮^⸮{⸮c|⸮|⸮⸮B⸮⸮ف20120528)
⸮cn;"f⸮⸮⸮c⸮?7⸮c⸮~v⸮"⸮⸮⸮c⸮2⸮⸮⸮⸮rtn3⸮⸮⸮⸮⸮⸮3
⸮⸮g+⸮⸮⸮r⸮g^⸮w⸮⸮⸮~⸮⸮⸮⸮>⸮kg⸮⸮a⸮6⸮⸮⸮⸮p⸮⸮⸮⸮⸮⸮  ⸮⸮⸮⸮v⸮fw⸮⸮⸮w3⸮⸮⸮⸮⸮f⸮⸮⸮⸮?⸮⸮v⸮ٻ⸮^:⸮s?⸮⸮⸮⸮0~⸮&⸮Mw⸮⸮⸮wwwn⸮⸮⸮⸮fn⸮⸮⸮⸮|g⸮⸮⸮v⸮⸮⸮⸮⸮⸮⸮1528000
⸮⸮ w"⸮⸮⸮⸮}rf⸮⸮⸮7gx^"7⸮}⸮v⸮⸮׳o:⸮m?ߗ⸮⸮w3_b~⸮,r⸮⸮⸮⸮⸮73?⸮⸮⸮⸮wuw3w|⸮֛⸮⸮⸮}s57⸮⸮⸮⸮w⸮⸮⸮⸮w;#⸮q3⸮2^Y3#^⸮/NtE⸮3ww⸮⸮n⸮3⸮⸮2⸮3⸮l⸮⸮⸮⸮rd⸮⸮3⸮⸮w⸮⸮⸮⸮R⸮t#}⸮yE~⸮gye~⸮g⸮'~⸮gyG~⸮g⸮G^D⸮,⸮D⸮⸮⸮E}@~~⸮~⸮⸮⸮⸮! w⸮w7ww⸮⸮v⸮ Freeing unused kmrn{[䫽⸮⸮⸮⸮⸮3⸮6⸮⸮>3<⸮lw3 {lv;r⸮lw;⸮⸮⸮⸮w⸮⸮e{⸮~⸮'{e~⸮~⸮׻⸮⸮⸮w⸮⸮׻⸮wߺ6Mf⸮L(bd⸮⸮~pg⸮⸮⸮~⸮~⸮⸮⸮⸮⸮⸮⸮⸮⸮ww⸮Ϳ⸮׻⸮⸮gN⸮⸮⸮o2g⸮&v'>{B~⸮N6⸮e}Btߺ⸮⸮⸮~⸮⸮⸮yyyyyyyyyyyyyyyy!y{Ayy{E~⸮{Eb⸮⸮⸮⸮⸮7⸮⸮⸮w⸮⸮⸮ggw⸮⸮G⸮?67?⸮⸮{⸮⸮⸮⸮7⸮⸮?.n⸮n2⸮⸮⸮o/⸮⸮⸮n2?3f⸮$⸮'77⸮⸮ww⸮⸮⸮⸮⸮⸮⸮⸮⸮f"D@Dd⸮f$f$&bDdffff f Fd&fFF&f$
```

nic3-14159, 2021-02-10:
Which software are you using to connect, and do you know the model of the USB to TTY converter you are using?

Guillermo-Ramirez-Jimenez, 2021-02-10:
First, I tried with Arduino IDE on Windows, then with Minicom on a Ubuntu VM. The converters are a PL2303 and a FTDI232.

I think it may be a config issue in Windows, data bits, parity, control flow, etc. I'm using the default options but I may need to change something.

nic3-14159, 2021-02-10:
The options I have are 8 data bits, no parity, and 1 stop bit (shows up as 8N1 at the bottom of Minicom). Maybe also check that the converters are set for 3.3V and not 5V? If those don't work, maybe try another terminal software, such as HyperTerminal or PuTTY

Guillermo-Ramirez-Jimenez, 2021-02-10:
Thanks but nope. I checked everything and tried PuTTY but I still get the same.

Here's the log:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.02.10 02:41:03 =~=~=~=~=~=~=~=~=~=~=~=

1.0.0.47 (De,¢ú²ÆšºÁ/ô1t367lçnw;Oôvwû~ÿfç~f÷ö÷gcÆvgÿv~ÿ÷fÿw~f?¶ßn¦w;î:$wûþö³þÿþÿ«™…Õ±Ñenvironment

In:    seriaì
JOut; ðHîOµç~ã£¶>·2ô¾¶¿vÿæwÿ¶û…~~w~ÿf÷÷ÿv?LÆÇ3ûf7lw/f'#þ¿v¾ãûoçôwo«xô–ÿöûÿöÿX‹Á, sector size 128 KiB
  Page skze ðH$I"Ö’¢å ÷vÇ3ûn³¿ÿ²~îvo&3ÿ¢FâO2ç¿þÿ÷·>>v2?vún2»~d²ã&wîgg>ç‘wo‘û&ÿÿþþ6öÁ00
17þ»ÿ2||¬÷g6odÝäz÷þþ·gÿ~ì²·ÿf÷»&þ¾wfw3þ|îgôLï“¦Ç2þöÿ ÿ>“ÂÂ‚åÑ•Íread: OK

NAND read: dgvic÷Hfï™»sÿ$ü~ç2ö|&?¾·¿6Oòÿ2þ~æ|üLö&÷2l·ww¶¿v¿>´õw:&ÿ÷fÿL&Â‚‚‚‚rrrj
   Image Name:   Linux-4.6.0k^ÛoË+?Ç$r/¿çg?òçþöâöþ¶÷s¶~»ú$w»wûöÿv{¾÷#þNf'w2off|·»v~l÷/¹÷:dLw2þ~w¾~v_ÏöTree blob at 02000000
   Booting uséng thå ³Y7Oþfô~úfû2º|&·v&vÿoxVûgþÿÿÿ~÷d~vþ¿ÿ&æû	2Í03b90 ... OK

Starting kernel ...MÇÃo¯oÿö÷÷öþersion 4.6.0-xilinx (root@alvin) (gkc »Yî¯½»Á7·&;"ü~³¶üþ:7ó÷ïwnwú¿²~þÿÿÿ¶|<w³|ïºç7fw³6w¿|¾:g´$ÿvn7Í¿¿cþ~öÿ~öÿçþÕÑ¥½¹cache
Machine model: xlnx,zynq-00øGá­ë"v{¾¶ÿg{´÷"O~>öþf?w¦Çrúþþÿÿ&7o7þ»û&7¦ÿ,¶¶w"÷ýÏ²w#tål¤þÿÿö~þ®«Á¥¹on.  Total pages: 65024
Kernel coíoaïdôm¿>ô¿»ÿ³¾þ>7ûî&f ~ÿþÿf÷þÿ÷ÿ~f>3vv¶÷"þo&ç;ÿ~ö÷g¿v7g÷îÇ??÷fçÿ•Éé 5, 131072 bytes)
Inode-cache hash tx,»O:¿vvzÿ÷2w3¼ÿfûó³>çw·sw›öÇföÿ÷ç?$÷çþ~f36>Oö;ï±çr»ûdowö¤w3wúöçöÿóÿ¥¡µ•µ¥
Virtual kernel memory layout;-N! `!veót&´76?¶ÿfþlw¦³nf·Lw2lþfç~ÿþ>?ÒÆç3º|ä;âóÿf¼|&&7?²FÇ7ÿ&äj÷nfçfvÿHB’ª²j	¥
    pkmap   : 0xbfe00000$kHfÍÑØL72¼|dv³ülfvg»æfç3¶³µÏ"~w~vf>oÏ2wö?t¯äçs&~L¶onz¿ö"v~>w24Lf7ïfÏþþ~w~‚ÂÁ900000 - 0xc0939e60   ( 232 kB)
J  `  `0.·¾w"æ~v·æ?þäç&¿öÿ37¿3vOf>÷~þL»Å~û~7ÿw·wþn÷?ºöþv'÷þ'öõsþ¾ûþö¶s=2.
RCU: Adjusting geometry for rcußfanuþÅ·c÷ofw ÷¾¿Çw~öw2¿æû{¿tnÇvþÿ÷ÿþÿÿ${f{o{~vÿ~~¾û»¿þfçg~~wKùÇ2¼“ÿvþ~wÿ~þõmodifies aux control register: 0x“š¶‚‚†Á ûÆ¿·²çn7òw¦v¿ç3þög¾];ï²?þ?öÿ»7·ïo¶·bö6ÿÿN÷tû~n·Ù¼'>õw~Lÿöþþÿþÿ1ÉC-310 dynamic clock gating enabled,$>Wö™ç mÏ—çg¿|lûn¾Ìû?Ç2²¿îgô¾û÷~ÿÿv¾|®úN~¿þoo~ogòÏ½ÿ"óãr¾nvôô»þ~÷~÷þk: 64 bits at 333MHz, resolution 3ns, wraðs`õv×ôMwgwºî¦þ7ÿsÿÿoowëÙÿÿÿ~þÿÿ~ÿçÿï0þOî··w»î²Fÿwó³ÿ~÷·dÿ$¾~÷þÿþþHh±½­Í½ÕÉ•é ttc_clocksource: mask> 0|fff÷HÛxÿ=o÷÷d¾~¶öæþ~¿ïÿûçßíbþ÷~÷ÿw¦t§~o÷¾vûno2{#lïÑç:ÿût¼]vÏ&ÿgö~ÿçö÷ÿ¹ 666.66 BogoMIPS (lpj=3333330)
pid_max:`ôgfñwö§3væ§ÿo÷ÿ÷~ö&wv?w÷÷ûçÿöþwÿ>l»$ût¿~÷'ä&fLç{fû>gvò§£ÎÙ$÷~wÿwç~ÿÿþÿkåé ok
CPU0: thread -1, cpu 0, socke| 4ÖHÛûdýdÇ2¼|f7?çº6ÿoûb>äct~lfÿvÿÿÿÿ×ï2tþæw|2ÿwv~tü&ö/÷?N÷GëÁtÏ•öÿfÿ~ÿçÔë½5%AM¥.
CPU: All CPU(s) started!in!sVÃ [Ï—ýç¶÷¼wî¿w~û¿vòöÆtôÿÿÿÿþ÷{2·~¶vÿï"?s2¾~·ÿf÷f÷L§çg~l2w÷÷Ù2750000 ns
pinctrl core: initializmd ¸^{ÏÙ³"w¾wó¾OæOoöw?óv t¾ógÿf~ö~òwû¾û'v¾ïdïv{>ûod{nG:ï½Ïro~ÿÿ÷;öÿ®+Ë‹and 1 watchpoint registers.
hw-brgaëpoént'ä?zûÿûlwÿwö¿§vú¿ÿ}äodÿfÿÿóþ>‘ÿ3þ¹ûc¾?ûçzÿsö~o{vÿäûoñåcûÿþÿ¿J¹¥Ñ¥…±¥é•‘5
usbcore: registered new z‹·É™§Í£ævzgbwòf?>öïÙ3ÿggznOÿf÷÷þ~§÷|ö6¿~~þ÷?gû7öÇ.þLs5o÷]çÿÿþþore: LinuxPPS API ver. 1 registered
pps_coùYƒsût>î»>´ÿÿ6öóþrÞ¾ívvþfwþÿ÷ær>vn¿~Ù»´?¢ÿ3O>Gÿãnïd|•¿÷|hÿÿþöþþþÿþ—C¡,ëk­Í½ÕÉ•é Switched to clocksource`armßglïbzó¿·ßs'öÿgû>wg¶d÷ÿúç¿&÷wÿææOrô¾o¿$þ~n;v|ûn¶ÿvúnÃ2¼OgOþÿÿgöfçes configured (established 2048 binl 2ØMç©äîFt|ÿözd<¿>vÿvûû{ö7¿~§{f~þ~çÿ÷;úºöowþ{Ûdþ¾ócÏd÷ü;«ôæZLE/:d÷÷þÿÿÿt module.
RPC: Registered udp transrorv`íodýleúvOp~7ÿïú>wûgön÷>~¿¾kp{þþÿÿç¿ÓÏ:f{gÛäwôü~~÷ö¿Ïwe~Uç"ìç»ÿg¶/B…Í¡table entries: 512 (order: 3, 327ÿNdVt{®w§÷ozgÿt"t{gþg{ÿ»{t¿ö?öþÿæÿLf6î¿væ~w{û»Ÿno¿¶ÿgö·ò ·o¼HgöZ.W–•‘5
io scheduler cfq registered (dYVì~tjGºï÷ò¾3þ~>ç2w2ú··Çs\¿ÇfÿfÿþÿÇô~ÿ~ö÷gûo{&ÿo÷g·þt?7›Gºrvþ~÷ÿÿÿÿþ&‚BJÉÅ= 166, base_baud = 0) is a`uar|liýgM
güLfÇ0¾º·¾ïÿ:³üþtwæ~·».ÿ§f÷fþÿ7#gæº¿:3~&þfws³~/w2Ç3tOFÏcwþwþþ±•[ttyPS0] enabled
xdevcfg g8p17p10·äg¿>ö2nöûvÇ6¾|üNwv7o´}ÿöçfbrd: module loadgd
ºl:2¿ÿ{ÿ&¶¿-ýrú÷ÿÿvþÿöid GPIO
libphy: MACB_méi_âuwÝHÿ¾™÷æOüýg${l7tün·í_nÿöÿþ?ûtçw?æ;7¦çãÿN~þf÷2»ü~÷z~eÝf'ä~v~ÿfþÿÿBj¥¥}‰ÕÍéphy_addr=e000b000.etherne:00l0kvq?1+cóûnfü¶Çÿ6wg¿ÿÎÅ0üL÷ÿç¿ççloçód6ÿòf>4|ÒôO•»cer÷³þþþÿ®¬K*!
%PCI platform driver
usbcK•ëÁr[×{Ñû?vÿg¤ûÿgÿg¯föþûÿv|]÷ïþÿÿ~ÿy6<H™ïÙÇg÷î»wôîûg¿gçv¾{ç÷~öv¬;c÷N³Ïfþvþÿþ÷_.®jŠŠ
‘‘Éõ0e, data=03
rtc-pcf85›&dS	†Õ3¿HîÏ<Ó÷'vOgÿä·û~ìþöoÿvÿvg{ÿÿ~ÿþÿçöfç7ÿl.vþLf¦vE÷þþþXilinx Zynq CpuIdle Driver started
Jsdhãk~1w{¾·d¦¿vÿgþÎ¿Ÿ7÷ûvÏlû~ÿþÿþÿæÿô|›çÿçþ·çvö#öã÷r{þoº¿|>~:óûtvÏ³÷r°¿þ»~w~÷¹DMA
ö6òkÿ>çÿþ÷þûç¿~®öoÝógÿvÿgþÿgÿusbhid: USB HID coru drkvõrMö¾woç÷»þfæ:ÿç¿÷ÿ~vÇ¿ôf÷fö|6vþö~÷ÿnand: Unknown W29N01HV
núdÿLf'î~fþN7zç—öþ¾7ooÿyNfÿæÿ÷þ|?f»ö~l|îögö7þ’fÇvsûÏ?âãÇþBad block table found at pagm 6=40ÜOdgr¾ïûO¢ï2wgç~ûäoú?·Åfzlÿÿÿ'›÷:ö¿n?s¶g?7ç2¾|fç2gpL&Gçþc÷‚ÂÁ00007fc0000
9 ofpart partitions f¯¹³Áo÷LÓg÷çv~¼vOl7¾gæO³zÿ~wÏ|çû÷þ÷ÿ¿¾¦vüw~wpù0üLf&çf3N¦?w&wfû7wå ¿Lÿÿÿf÷0x000000820000-0x000001ù&fp0´'fÏg÷ûÿoþw÷rÿt~f~w~÷~÷ÿ³.&¤Høÿw|w™wç~ö7§ÿä{¾o~·ÿvú{çw·&ÿkôg¿fþ~sit: IPv6 over IXv4(tu÷•³å¹¿"dûv{¾ÚþÞGêgû÷gz.wò~fþÿÿÿ?wóï2ÿö{vg¶~n|özv¿¶~dwúvcï>½±(rev 20120528)
ãc·'dúvvûg÷|vfþ¿óöÏ‘´¾þ²ÿfÿÿþ?ó?~vÿÓg?ÿÿþ?ÿÿøc¾w~~Çþû?»¢¿bo~÷~ôl¿¤÷ÿ~ö¾ö÷üo‘»?Nÿÿ~ö÷÷ßf·~äg»ïwwwvf&DDFf6f$FDFff&$DfFF#$$DFDFDDÄDffffffffffffff`ffffffæf` æfff!-–ÖÉ: Empty flash at 0|°156¢Þƒ"n~d"gf327þ~fw"3&$&nFDDD$,fffffæÿÿÿ~ï>ovö~oüó&'>æÿdæl&fÇ3_&f0ºçffffffffffffæfææfæffæfææffPé™›ÍÉ~0GmðtL6?wúî>&w¶÷6ãv3»~ö.vd~ÿþþ·wb#2d$dFffBffddB&&$&FDFd2$dDDDD",Lfffffffffffææffffff`fff` f~f@@@@@@@@@@DDDDDDFFf3&dFlf&6f&ffffDD,LffffvþwwvfDDDff6&$FddFd&d,DfDF6DdDDFF@DDDLfffffffffffæff`f~fææffæff`ff @@@@@@@@@@ DFD$FddFdd&f$$DF6"&3dDDDLfffffægfc2$Dfffd&Fdfdf6dFf&df$ [... cut ...] æffffææffff`æfff@@@@@@@@@ DDd$ddfFDff3"$lFdD"s&&DÄbfffffþwfæ'LF&f2.fdfb0$$fdfF4&dFFDF$DDDDbffffffffffæff`ffffæfffff@@@@@@@@@DDDDDff&3d"$Fff$3ddfFffDDDLLfffff÷æg'3&ddlfff&6d$bfFf6&&fDDDdf$DDDDDffffffffffffæff&fææfæfæfffff@@@@@@@@@@DDFDd6dDDf&$&dFDr"3&6fDDÄDf&ffffÿöögr22$ddnFF&"&dDfFff4$fDFDdF$DDDDDDfffffffffffffæfff~ææffffæfffæ @@@@@@@@@@DDDFF &dd&dDf2"dDdfff66DDDfþf

PetaLinux 2016.4 (none) /dev/tt}ÐS4c£¸LÔF~oçôÿûÿ&÷ÿó÷

PetaLinux 2016.4ÐNûÏ•§ »çÿñþ~þw?§göw:ÿ~?sÿï¥ÿ>þ÷÷
```

Guillermo-Ramirez-Jimenez, 2021-02-10:
Ok, I found it. It was my fault, one of the welds detached from the board and the GND pin was loose (almost unnoticeable) and that caused all the issues.

Everything looks fine now:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.02.10 15:40:39 =~=~=~=~=~=~=~=~=~=~=~=

1.0.0.47 (Dec 07 2017 - 00:36:15 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
NAND:  128 MiB
MMC:   zynq_sdhci: 0
Invalid bus 0 (err=-1)
*** Warning - spi_flash_probe() failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Net:   Gem.e000b000
Hit 'd' to stop autoboot(eric):  3  2  1  0 
Copying Linux from NAND flash to RAM...

Device 0: nand0, sector size 128 KiB
  Page size       2048 b
  OOB size          64 b
  Erase size    131072 b
  subpagesize      512 b
  options     0x       0
  bbt options 0x   20000

NAND read: device 0 offset 0x2220000, size 0x300000
 3145728 bytes read: OK
fpga_loadbitstream: Design name id not recognized in bitstream

NAND read: device 0 offset 0x300000, size 0x500000
 5242880 bytes read: OK

NAND read: device 0 offset 0x800000, size 0x20000
 131072 bytes read: OK
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-4.6.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3820688 Bytes = 3.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Device Tree to 0f2fe000, end 0f303b90 ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.6.0-xilinx (root@alvin) (gcc version 5.2.0 (GCC) ) #5 SMP PREEMPT Mon May 8 14:36:49 CST 2017
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: xlnx,zynq-7000
cma: Reserved 16 MiB at 0x0e000000
Memory policy: Data cache writealloc
percpu: Embedded 12 pages/cpu @cfdce000 s19776 r8192 d21184 u49152
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
Kernel command line: console=ttyPS0,115200 root=/dev/mtdblock6 rootfstype=jffs2 noinitrd rw rootwait
PID hash table entries: 1024 (order: 0, 4096 bytes)
Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Memory: 234624K/262144K available (5286K kernel code, 231K rwdata, 1848K rodata, 1024K init, 226K bss, 11136K reserved, 16384K cma-reserved, 0K highmem)
Virtual kernel memory layout:
[... 509 lines cut ...]

detach test_pool_thread 

detach test_pool_thread 

their_ipaddr =
open file error 13072 copyright_gettime
copyright_gettime error (13362 -- thread_CopyrightLoop)
open file error 13072 copyright_gettime
copyright_gettime error(13365 -- thread_CopyrightLoop)

..............................done

 [2018-10-22 18:01:55] Failed to resolve (?wrong URL) vip.okpool.me:443                    
 [2018-10-22 18:01:55] Failed to resolve (?wrong URL) vip.okpool.me:3333                    
 [2018-10-22 18:01:55] Failed to resolve (?wrong URL) vip.okpool.me:23                    
Start pools_active [2018-10-22 18:02:35] No servers were found that could be used to get work from.                    
 [2018-10-22 18:02:35] Please check the details from the list below of the servers you have input                    
 [2018-10-22 18:02:35] Most likely you have input the wrong URL, forgotten to add a port, or have not set up workers                    
 [2018-10-22 18:02:35] Pool: 0  URL: stratum+tcp://vip.okpool.me:3333  User: wuhai0202.12X31X91  Password:                     
 [2018-10-22 18:02:35] Pool: 1  URL: stratum+tcp://vip.okpool.me:443  User: wuhai0202.12X31X91  Password:                     
 [2018-10-22 18:02:35] Pool: 2  URL: stratum+tcp://vip.okpool.me:23  User: wuhai0202.12X31X91  Password:                     
```

Conclusion: check your connections thrice before asking.

closed 2021-02-10 by Guillermo-Ramirez-Jimenez