xjtuecho/EBAZ4205 #51: Communication gets stuck

rribalta, 2021-08-06:
I received my board a few days ago and now I am in the process of testing it but after connecting it to my computer this is the only I get:

```
1.0.0.47 (Dec 07 2017 - 00:36:15 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
```

It gets stuck right before printing the nand memory size, this makes me think it might be damaged since they wrapped the board with plastic that is easily charged electrostatically... Or I might be just missing something

Has somebody had the same issue?

firepower9966, 2021-08-12:
Check both sides of PCB, I had some resistor and caps torn off. I found value from schematic and replaced them. SMD size is 0402.

rribalta, 2021-08-16:
Thanks I will try that

rribalta, 2021-09-01:
Ok, time for a little update: I soldered a cap that fell off the board (it fell inside the box and I didn't notice), plus now I power it with 12V (before were 5V). Now the output is this:

```
1.0.0.47 (Dec 07 2017 - 00:36:15 +0800)

Model: Zynq Zed Development Board
DRAM:  ECC disabled 256 MiB
NAND:  128 MiB
MMC:   zynq_sdhci: 0
Invalid bus 0 (err=-1)
*** Warning - spi_flash_probe() failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Net:   Gem.e000b000
Hit 'd' to stop autoboot(eric):  0
Copying Linux from NAND flash to RAM...

Device 0: nand0, sector size 128 KiB
  Page size       2048 b
  OOB size          64 b
  Erase size    131072 b
  subpagesize      512 b
  options     0x       0
  bbt options 0x   20000

NAND read: device 0 offset 0x2220000, size 0x300000
 3145728 bytes read: OK
fpga_loadbitstream: Design name id not recognized in bitstream

NAND read: device 0 offset 0x300000, size 0x500000
 5242880 bytes read: OK

NAND read: device 0 offset 0x800000, size 0x20000
 131072 bytes read: OK
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-4.6.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3820688 Bytes = 3.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Device Tree to 0f2fe000, end 0f303b90 ... OK

Starting kernel ...
```

rribalta, 2021-09-24:
Alright, time for another update and probably the last one, I kept trying and I found that it gives different responses at random before getting stuck, here are some examples that caught my eye:

> 1.0.0.47 (Dec 07 2017 - 00:36:15 +0800)
> 
> Model: Zynq Zed Development Board
> DRAM:  ECC disabled 256 MiB
> NAND:  128 MiB
> MMC:   zynq_sdhci: 0
> Invalid bus 0 (err=-1)
> *** Warning - spi_flash_probe() failed, using default environment
> 
> In:    serial
> Out:   serial
> Err:   serial
> Model: Zynq Zed Development Board
> Net:   Gem.e000b000
> Hit 'd' to stop autoboot(eric):  3  2  1  0 
> Copying Linux from NAND flash to RAM...
> 
> Device 0: nand0, sector size 128 KiB
>   Page size       2048 b
>   OOB size          64 b
>   Erase size    131072 b
>   subpagesize      512 b
>   options     0x       0
>   bbt options 0x   20000
> 
> NAND read: device 0 offset 0x2220000, size 0x300000
>  3145728 bytes read: OK
> fpga_loadbitstream: Design name id not recognized in bitstream
> 
> NAND read: device 0 offset 0x300000, size 0x500000
>  5242880 bytes read: OK
> 
> NAND read: device 0 offset 0x800000, size 0x20000
> **data abort**
> pc : [<0ff539bc>]	   lr : [<0ff53668>]
> reloc pc : [<0402a9bc>]	   lr : [<0402a668>]
> sp : 0f3054a8  ip : 00000000	 fp : 00000000
> r10: 00800000  r9 : 0f308ef8	 r8 : 00000000
> r7 : 00000000  r6 : 00000000	 r5 : 00000000  r4 : 0ffbaaa8
> r3 : b5e7f35b  r2 : 00020000	 r1 : ffffffff  r0 : 0ffbaaa8
> Flags: nZCv  IRQs off  FIQs off  Mode SVC_32
> Resetting CPU ...
> 
> resetting ...

> 1.0.0.47 (Dec 07 2017 - 00:36:15 +0800)
> 
> Model: Zynq Zed Development Board
> DRAM:  ECC disabled 256 MiB
> NAND:  128 MiB
> MMC:   zynq_sdhci: 0
> Invalid bus 0 (err=-1)
> *** Warning - spi_flash_probe() failed, using default environment
> 
> In:    serial
> Out:   serial
> Err:   serial
> Model: Zynq Zed Development Board
> Net:   Gem.e000b000
> Hit 'd' to stop autoboot(eric):  3  2  1  0 
> Copying Linux from NAND flash to RAM...
> 
> Device 0: nand0, sector size 128 KiB
>   Page size       2048 b
>   OOB size          64 b
>   Erase size    131072 b
>   subpagesize      512 b
>   options     0x       0
>   bbt options 0x   20000
> 
> NAND read: device 0 offset 0x2220000, size 0x300000
>  3145728 bytes read: OK
> fpga_loadbitstream: Design name id not recognized in bitstream
> 
> NAND read: device 0 offset 0x500000, size 0x7b00000
> **size adjusted to 0x7a60000 (5 bad blocks)**
>  128319488 bytes read: OK
> 
> NAND read: device 0 offset 0x800000, size 0x20000
>  131072 bytes read: OK
> bootm - boot application image from memory
> 
> Usage:
> bootm [addr [arg ...]]
>     - boot application image stored in memory
> 	passing arguments 'arg ...'; when booting a Linux kernel,
> 	'arg' can be the address of an initrd image
> 	When booting a Linux kernel which requires a flat device-tree
> 	a third argument is required which is the address of the
> 	device-tree blob. To boot that kernel without an initrd image,
> 	use a '-' for the second argument. If you do not pass a third
> 	a bd_info struct will be passed instead
> 	
> For the new multi component uImage format (FIT) addresses
> 	must be extened to include component or configuration unit name:
> 	addr:<subimg_uname> - direct component image specification
> 	addr#<conf_uname>   - configuration specification
> 	Use iminfo command to get the list of existing component
> 	images and configurations.
> 
> Sub-commands to do part of the bootm sequence.  The sub-commands must be
> issued in the order below (it's ok to not issue all sub-commands):
> 	start [addr [arg ...]]
> 	loados  - load OS image
> 	ramdisk - relocate initrd, set env initrd_start/initrd_end
> 	fdt     - relocate flat device tree
> 	cmdline - OS specific command line processing/setup
> 	bdt     - OS specific bd_t processing
> 	prep    - OS specific prep before relocation or go
> 	go      - start OS
> zynq-uboot>

At this point I am almost sure there's something wrong with the nand memory so I will leave this here, at least it might help someone in the future. Thanks firepower!