xjtuecho/EBAZ4205 #60: Bad blocks in nand flash. Coincidence?

jcdevel, 2023-10-22:
Hello Everybody,

I recently got an EBAZ4205 board. U-Boot reports the last 5 sectors of the nand flash as bad blocks:
```
zynq-uboot> nand bad 0

Device 0 bad blocks:
  07f60000
  07f80000
  07fa0000
  07fc0000
  07fe0000
```
I wouldn't give it too much importance if it weren't because it seems that all boards have exactly the same sectors marked as bad blocks. I do not believe in coincidences, even less in this one.

Does someone know if there is any particular reason for which all boards seem to have these bad blocks? Or are there lucky people who got an ebaz4205 with a healthy nand flash?

Thank you very much for your time and your support!

nic3-14159, 2023-10-22:
If I recall correctly the bad block information is essentially just a file that uboot maintains. So if the same NAND image is copied to all boards they would list the same bad blocks information.

jcdevel, 2023-10-28:
I see. Thanks for the clarification! So I should be able to recover and use those bad blocks using U-Boot commands and store data of my own, right?

closed 2023-10-29 by jcdevel